abr/abr_control #19: thanks + pointer to other robot control library

ahundt, 2017-10-24:
@studywolf I've been a reader of your blog for quite some time and first I wanted to thank you for all the detailed and informative posts you've written. I saw your recent post about the motion control library. https://studywolf.wordpress.com/tag/vrep/

I wanted to point you to another set of libraries that go together which I've been using and have found to be quite good for similar purposes:
https://github.com/jrl-umi3218/Tasks
https://github.com/jrl-umi3218/SpaceVecAlg
https://github.com/jrl-umi3218/RBDyn
https://github.com/jrl-umi3218/Eigen3ToPython

This is really just an FYI which incorporates other interesting control approaches and also has been integrated with v-rep in various projects. Hopefully this is more useful or interesting than disruptive, but feel free to close. Thanks! :-)

studywolf, 2017-10-25:
Hello @ahundt! Thanks for the links, looks like some great work!

I remember briefly looking into different math libraries to use, do you know if there are speed benefits to using the Eigen3 library (wrapped) over NumPy / where I can find timing runs information?

ahundt, 2017-10-26:
I don't know of any specific eigen + cython benchmarks, so I can't give you details on that exact combination. However, there are quite a few general comparisons of cython vs numpy code which are very favorable for cython generally.
http://notes-on-cython.readthedocs.io/en/latest/std_dev.html
https://jakevdp.github.io/blog/2015/02/24/optimizing-python-with-numpy-and-numba/

As for eigen C++ library performance, eigen3 is what was chosen as the backend API for much of TensorFlow, and I've been using eigen3 for many years in robotics. As an anecdote, when some of my fellow collaborators worked on beating performance of a specific eigen call in C++ which was a bottleneck on an x86 based platform, they ended up just using eigen as-is.

studywolf, 2017-10-26:
Ah, interesting. I've definitely found the same regarding Python vs Cython, so we use Cython here for all of the config calculations, I'll have to try to find some time to do some comparisons with the Eigen3 version!

Thanks again for the above links, I will definitely be looking through the repos / playing around with the code! :)

ahundt, 2017-10-26:
Awesome! I'm glad you found the info valuable. I'll close this for now since it is slightly off-topic for your library.

closed 2017-10-26 by ahundt