abr/abr_control #29: ImportError: No module named joint0[0,0,0]_Tx.wrapper_module_0

cambel, 2018-11-29:
Hi there,

I have been following your work recently (Play around with the sympy lib and a UR3 arm). Now I was trying to use this abr_control library, I made it work with the UR3 model and the first time it runs kind of fine. I take about 10 minutes computing every... kinda slow but anyway it works. But when I try to run it again I got an error about loading a saved function. I tried with the pygame example to see if it was not a problem with my custom model, but I got the same thing after trying to run the example a 2nd time:

`Traceback (most recent call last):
  File "reaching_sliding.py", line 30, in <module>
    interface.connect()
  File "/home/cambel/dev/abr_control/abr_control/interfaces/ipygame.py", line 87, in connect
    self.arm_sim.connect()
  File "/home/cambel/dev/abr_control/abr_control/arms/threejoint/arm_sim.py", line 54, in connect
    self._update_state()
  File "/home/cambel/dev/abr_control/abr_control/arms/threejoint/arm_sim.py", line 121, in _update_state
    self._position()
  File "/home/cambel/dev/abr_control/abr_control/arms/threejoint/arm_sim.py", line 109, in _position
    for ii in range(self.robot_config.N_JOINTS)]
  File "/home/cambel/dev/abr_control/abr_control/arms/base_config.py", line 362, in Tx
    self._Tx[funcname] = self._calc_Tx(name, x=x)
  File "/home/cambel/dev/abr_control/abr_control/arms/base_config.py", line 754, in _calc_Tx
    Tx, Tx_func = self._load_from_file(filename, lambdify)
  File "/home/cambel/dev/abr_control/abr_control/arms/base_config.py", line 188, in _load_from_file
    filename + '.' + saved_file)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named joint0[0,0,0]_Tx.wrapper_module_0`

It seems to be unable to find the saved function.

studywolf, 2018-11-29:
Hello! 

In the example script, when you create the `robot_config = Config(use_cython=True)` if you change `use_cython=False` you should see significantly faster start up (because it won't be generating C functions for each of the calculations), but the execution time may be a bit slower. Probably that's fine though.

I believe that the import error is due to using Python 2.7. If you download and setup Anaconda, then it should be very quick to get set up in Python 3 running `conda create -n myenv python=3.5`. From there it should be quick to try running the examples and hopefully when you use `use_cython=True` there aren't any issues. Actually, I find that when I'm using Python 2.7 I'm not even able to run the example a first time without an import error. I will make a note that Python 2 isn't supported, thanks! 

Regarding building up a config for the UR3, awesome! Would love to get your feedback on the process, and if you have any questions or we can help with it please don't hesitate to write.

So, tl;dr: I think the import errors are cause by Python 2.7, if you switch to Python 3+ those should go away. Alternatively, setting `use_cython=False` in the scripts your running when creating `robot_config` might also do the trick. Please let me know!

studywolf, 2020-01-13:
Sorry about the crazy long response time, thanks for the example! We've discussed getting a DH parameters setup going for the repo, but are just short on cycles to develop it. If you are able to make a PR for it we would definitely appreciate it as it's super worthwhile to have!