abr/abr_control #42: How to create manipulator config files

fhln, 2019-04-25:
Hello, I have a new robot config information with urdf format file and D-H parameters, I wish to know how to convert it into config files of this project? The variables in config.py is confusing, can you tell me how to calculate them?

studywolf, 2019-04-25:
Hello! I have a blog post describing how to generate the transform matrices as they're set up in the config file here: https://studywolf.wordpress.com/2017/06/09/deriving-a-robots-transform-matrices/ If you have a VREP model of the arm it's especially straight forward. 

I'm happy to answer any specific questions about the variables in `config.py`.

At this point we haven't looked into integrating urdf files / D-H parameter descriptions yet, but I've added it to our backlog and we'll try to get to it in the near future.

bsangiovanni, 2020-01-04:
Hi,
apologies for bringing this up again, but I have some doubts about the `config.py` for the V-Rep UR5 arm.
When computing the matrix L
`        self.L = np.array([
            [0.0, 0.0, 1.4650e-02],  # link 0 offset
            [0.0, 0.0, 8.5001e-03],  # joint 0 offset
            [-7.1771e-03, 1.1159e-04, 7.0381e-02],  # link 1 offset
            [-6.3122e-02, -9.5099e-05, -4.3305e-03],  # joint 1 offset
            [2.1255e-01, -9.9446e-04, 6.4234e-02],  # link 2 offset
            [6.4235e-02, 1.1502e-04, 2.1255e-01],  # joint 2 offset
            [1.8677e-01, 6.7934e-04, -5.7847e-02],  # link 3 offset
            [-5.7847e-02, -1.6153e-05, 2.0538e-01],  # joint 3 offset
            [-7.5028e-03, -5.5328e-05, 3.2830e-02],  # link 4 offset
            [-6.8700e-03, 4.5318e-05, 5.3076e-02],  # joint 4 offset
            [3.6091e-03, 5.0090e-05, 4.2340e-02],  # link 5 offset
            [1.0824e-02, -4.5293e-05, 6.8700e-03],  # joint 5 offset
            [0, 0, 7.6645e-02]])  # link 6 offset`

I can see from V-Rep that the data inserted for the joints are correct, but I can't seem to find any consistency with the numbers you used for the joints. How are these computed? If I follow the procedure in your blog with the default UR5 model, they don't compare.
Thanks for your great work, if you could help me clarify this it would be much appreciated!

studywolf, 2020-01-06:
Hello! 

So just rechecking these numbers myself, the smaller ones (anything less than 1e-3) tend to vary a bit, but all of the larger numbers are consistent. If you're seeing discrepancies with anything around 1e-2 or 1e-1, can you double check that you've selected relative to: 'Parent frame' when looking at the offsets? 

For example, looking at the link 3 offset I see
```[1.8677e-01, 1.0192e-5, -5.7847e-2]```
so while the y offset is different than that in the config file, it's also at a small enough scale that the effect is negligable (also if you simulate the arm forward for a short while and check again you'll see that the reported y offset has changed again). 

Does that help? Or is there a different source of inconsistency?

bsangiovanni, 2020-01-08:
Thank you so much for your reply! You're right, the larger numbers are consistent, while the varying ones are in very low order of magnitude: I figure an error of 1e-05 m is probably not as influential in this case.
Thanks again and props for the great work, I am studying robotics and I find all this very helpful.
All the best

studywolf, 2020-01-08:
Ah, glad to hear it was just the small numbers. Thank you for the kind words and best of luck with your studies!

closed 2020-01-08 by studywolf