abr/abr_control #57: The vrep scenes with Jaco2 have six joints, but the abr_control uses only five. This is expected?

juanjqo, 2019-09-15:
Hi!, your repository is amazing. Thanks for sharing!

I realize that the configurations space of Jaco2 arm from abr_control module is a vector 5x1. However, in the vrep scene, Jaco2 have six joints.

```
from abr_control.arms import jaco2 as arm
from abr_control.controllers import OSC
from abr_control.interfaces import VREP
import numpy as np

robot_config = arm.Config(use_cython=True)
ctrlr = OSC(robot_config, kp=20,
            # control (x, y, z) out of [x, y, z, alpha, beta, gamma]
            ctrlr_dof=[True, True, True, False, False, False])
interface = VREP(robot_config)

interface.connect()

target_xyz = [.2, .2, .5]  # in metres
target_orientation = [0, 0, 0]  # Euler angles, relevant when controlled
for ii in range(1000):
    feedback = interface.get_feedback()  # returns a dictionary with q, dq
    q = feedback['q']
    print(q.shape)
    u = ctrlr.generate(
        q=feedback['q'],
        dq=feedback['dq'],
        target=np.hstack([target_xyz, target_orientation]))
    interface.send_forces(u)  # send forces and step VREP sim forward
interface.disconnect()
```

Output:

> Connected to VREP remote API server

> (5,)

Expected:

> (6,)

Why q = feedback['q'] returns a vector 5x1 if the robot have six joints?

Thanks for your time!

Att

Juan

p3jawors, 2019-09-18:
Hello Juan,

Thank you for the kind words! When you instantiate your robot config, you need to specify whether you are using the arm with or without a gripper. If you change your first line after imports to

`robot_config = arm.Config(use_cython=True, hand_attached=True)` it should fix the problem.

What's happening here is that when we do not have the hand attached, there is no link after the final joint, so we do not include it in the controllable joints. This changes the feedback from the arm to be of length `(5,)` instead of `(6,)`.

Hope that clears things up, let me know if it doesn't solve the problem.

Cheers,

Pawel

juanjqo, 2019-09-18:
@p3jawors Thanks for your help!
It is working as you mentioned!

closed 2019-09-19 by p3jawors