abr/abr_control #60: 3D model files missing

xuhuairuogu, 2019-10-15:
Hi, I am a new user of abr_control. The examples in the directories of PyGame and VREP both work well. However, when I execute `python3 dynamics_adaptation_osc.py`, I receive the following error:
[image]
And indeed there isn't any 3D model file in meshes directory. Where can I get jaco2's model files?

xuhuairuogu, 2019-10-15:
I download model files from kinova-ros package and it works!

studywolf, 2019-10-15:
Excellent! Glad you found them and it's working for you. We decided to not include the stl files, but as you found there's a comment in the xml file about where to download them. I'll make an issue to add that to the readme so it's easier to find. Cheers!

closed 2019-10-15 by studywolf