abr/abr_control #61: AttributeError: 'MjViewer' object has no attribute 'exit'

superggb, 2019-10-17:
I have installed mujoco200 and mujoco-py2.0.
But when I run the /example/Mujoco, there is an error in this code:
'if interface.viewer.exit:'
           .......

So I can't run any Mujoco examples for all they have this code in the first 'While'.....

I have searched it online, may it be the prolem of version of mujoco-py ?
For there is not mujoco-py‘s and mujoco's version mentioned.

studywolf, 2019-10-17:
Hi SuperGGB,

Whoops. I made some changes to mujoco-py and forgot to make them public. I'll fork the repo and post when there's a branch available here to use.
Cheers,

studywolf, 2019-10-17:
OK I have the change up on a branch here https://github.com/studywolf/mujoco-py/tree/exit_externally_handled

You can either clone and change to this branch or you could look at the changes from the last commit and apply them locally. Please let me know if this doesn't work for you!

superggb, 2019-10-18:
yep! this solution worked.
Thank you a lot !

closed 2019-10-18 by superggb