abr/abr_control #68: error occurs when calculation of 'C'

vanusy, 2019-12-31:
When I run file ' position_control_sliding.py' to test, an error occurs as follows:
'UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc3 in position 908: invalid continuation byte', which is in file base_config.py Line 769 in _calc_C. 
I compare the calculation of J, M, G, it seems that only calculation C is nor right.

studywolf, 2020-01-06:
Hello! 

Are you running this in the PyGame or VREP examples? I'm able to run this example on my computer. Can you provide more details about your setup? One quick thing to try is deleting the `~/.cache/abr_control` folder, which will force the cython functions to be regenerated.

vanusy, 2020-01-07:
Hello! Thanks for your reply.  I have ever read your blogs on 'https://studywolf.wordpress.com/', which gives me a lot of help when I study robotics. Thanks again for your effort.

The problem is detailed as follows:
1. The abr_ontrol project is running on my PC 'windows 64' and 'anaconda3' with 'python 3.6'. All the dependent  libraries (NumPy, SymPy, SciPy, CloudPickle, Cython, SetupTools, Nengo, and Matplotlib) have been installed. 
2. When running the file 'abr_control-master/examples/VREP/position_control_sliding.py', error occurs:

Traceback (most recent call last):
  File "E:/python-prj/abr_control/abr_control-master-20200106a/examples/VREP/position_control_sliding.py", line 52, in <module>
    target=target_xyz
  File "G:\anaconda3\lib\site-packages\abr_control\controllers\sliding.py", line 88, in generate
    C = self.robot_config.C(q=q, dq=dq)
  File "G:\anaconda3\lib\site-packages\abr_control\arms\base_config.py", line 337, in C
    self._C = self._calc_C()
  File "G:\anaconda3\lib\site-packages\abr_control\arms\base_config.py", line 766, in _calc_C
    parameters=self.q+self.dq)
  File "G:\anaconda3\lib\site-packages\abr_control\arms\base_config.py", line 146, in _generate_and_save_function
    args=parameters, tempdir=folder)
  File "G:\anaconda3\lib\site-packages\sympy\utilities\autowrap.py", line 632, in autowrap
    return code_wrapper.wrap_code(routine, helpers=helps)
  File "G:\anaconda3\lib\site-packages\sympy\utilities\autowrap.py", line 147, in wrap_code
    self._process_files(routine)
  File "G:\anaconda3\lib\site-packages\sympy\utilities\autowrap.py", line 170, in _process_files
    " ".join(command), e.output.decode()))
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc3 in position 908: invalid continuation byte

VREP connection closed...
Simulation terminated...

3. I check it step by step and find 'base_config.py', the '_calc_C' function in class 'BaseConfig' Line 764  is the bug. While other functions '_calc_dJ' and '_calc_M' are correct.

I'm confused about it. Could you give me a help?

vanusy, 2020-01-07:
Also, the cache files saved by the first computation have been deldeted. But the errors still occur.

studywolf, 2020-01-07:
Hmmm, I haven't encountered this bug before. I left the program running overnight and I was able to successfully generate the `C` matrix on my Ubuntu 18.04 system with Python 3.6.0.

Could you confirm for me whether or not it also happens when you run the `PyGame/position_control_sliding.py` file? 

These are the versions of the dependent libraries I'm using, can you check against the ones you have installed?
```
numpy == 1.17.1
sympy == 1.4
scipy == 1.3.1
cloudpickle == 1.2.2
cython == 0.29.13
setuptools == 41.0.1
nengo == 3.0.1.dev0
matplotlib == 3.1.1
```

vanusy, 2020-01-10:
Thanks for your detailed information about the program environment. I have conquered this problem with ubuntu os and python 3.6. I'm so glad that I can develop my research project.

studywolf, 2020-01-10:
excellent! glad to hear :D if there's anything else we can help with please let us know!

closed 2020-01-10 by studywolf