abr/abr_control #74: What is the effect of dt in your expressions?

quartzolit, 2020-02-28:
I can see on your examples you set the dt value in order to connect and set time step of the simulation. But when I designedly increase the dt value, higher is the joint torque value (u) and consequently the control fails at higher dt value. However, no step time is used on your equations, besides when you set up the connection to CoppeliaSim.

My main objective by asking this is to use your repository via ROS, and controls UR5 robot arm at CoppeliaSim using ROS interface. By doing this it is not possible to force a time step, and both (CoppeliaSim and ROS) will run asynchronously. 

Is that a way to control according to dt, or I need a powerful computer to run both as fast as possible, in order to indirectly generate a lower dt value? (Also thinking in real life robot arm control)

studywolf, 2020-02-28:
Hello! Basically the `dt` sets how often the control signal can be updated. The dynamics are estimated every time the control loop runs, and for systems with highly nonlinear dynamics the control signal generated is only relevant inside a small area of state space.

So if you generate a control signal `u_t` when the system is in configuration `q_t`, as your robot gets farther away from `q_t` the control signal `u_t` is going to be more and more outdated and cause undesired behaviour.

If things are running asynchronously and you can't control the `dt` (of the sim or in real life) then yeah, you're going to want to have a good computer to calculate the control signal as quickly as possible. The UR5 control signal generates pretty quickly if you compile the cython functions in the config file (which is default on the repository), but unrelated to having a good computer to calculate the control signals, you also really want to watch out for the communication overhead. We found this to be the primary issue when communicating with the Jaco2 arm, and ended up having to build our own communication protocol with it (github.com/abr/abr_jaco2) to minimize it.

Tangential question: Does the ROS interface with CoppeliaSim have a force control mode? 

Secondary tangential question: Does the UR5 have a force control mode? We had looked into it a few years ago and there was no real option for controlling the torque at the joints.

You could also use the path planners, if you're doing position control with the UR5. There's an option with the path planners to track the wall clock time since the last time it was called and send out a target position based on where in the trajectory the robot should be (e.g. we're 2 seconds into the 4 second movement, the robot should be halfway through the trajectory)

quartzolit, 2020-03-01:
Thanks for this quick reply. This information came handly.

Now Answer your questions. First about ROS.

ROS Interface doesn't have a proper force control mode. This interface only give us the possibility to send data between ROS and Coppelia (using publishing and subscribing concept). So this provides only a data link, similar to the real world. In other words, V-rep only gives me the q and dq feedback, then I'm using your repo to force control generating torque vector (y), and finally send it back to v-rep to apply those changes. This gives me a similar relation to control the real arm.

And about Ur5. I can see on its user manual it does have a force command, which I can imagine it is possible to control using this mode. But until i did not resolve this control task using simulation. I cannot test it for real. Also UR has a ROS package, it might help

studywolf, 2020-03-05:
Ah I see, so if it's the same as it was when we were using the UR5 then it has some sort of 'apply a force at the end effector option' that you can use through their interface. If you find a way to apply torques to the joints with an update rate better than 100Hz please let us know!

quartzolit, 2020-03-06:
Sure!

quartzolit, 2020-03-08:
What's a good computer requirements for a good asyn simulation?

closed 2020-08-15 by studywolf