abr/abr_control #75: Can I download those mesh and texture files directly?

HeegerGao, 2020-03-04:
Hello, I want to run some examples in Vrep, and I have to 'Download mesh and texture files to run sim'. But I am in China, and I always get a 'Connection time out' error when I am trying to download them.

I want to know if you can give all the links of the mesh and texture files so that I can download beforehand(it is more convenient for me because I could use a lot of network tools to download them), and also could you tell me where to put those files if I download them by myself? Thanks.

HeegerGao, 2020-03-04:
Or maybe you can directly upload them in this github repo if they are not quite big.

studywolf, 2020-03-04:
Ahhh interesting, this scenario had not occurred to me. They're big enough that I'd like to keep them out of the repo, but we will figure out an alternative place to store them you can access! In the mean time, if you email me the specific model that you're trying to run I can send it to you directly: travis.dewolf@gmail.com

closed 2020-03-05 by HeegerGao