abr/abr_control #76: Other arms with ABR_Control

animesh-garg, 2020-03-10:
Currently, the models support OSC controllers for  Jaco and UR5 
Is there a plan to do this for other robot arms, Sawyer or Franka Emika. 

Further doing the same for these arms in Coppelia and Mujoco?

studywolf, 2020-03-10:
Hello! 
We don't have any plans to add support for more arms at the moment, but I do have a blog post that walks through the steps of generating your own config files for any Coppelia model: https://studywolf.wordpress.com/2017/06/09/deriving-a-robots-transform-matrices/
If you go through the process and would like to contribute the config file to the repo that would be great!

For Mujoco, any robot arm is supported as long as there is a `<site>` named `EE` attached to the end effector of the robot. If you have the XML and mesh files for a model and would like to link me to them I'd be happy to go through the process for getting it working with ABR Control to generate a formal set of steps for people to follow.

studywolf, 2020-08-15:
Hello! I'm going to close this for now, but if you'd like to discuss further we can re-open!

closed 2020-08-15 by studywolf