abr/abr_control #80: xyz_offset argument with problems

quartzolit, 2020-06-12:
I've added an ur5 arm on a mobile robot. Because of that i need constantly update its base position. But when I use the OSC.generate and robot_config.Tx with the specific xyz offset. It returns a different value that is supposed to be. For example:

´´´
base = interface.get_xyz("UR5_link0_visible")
result: Base is:  [-9.712021827697754, 1.791999101638794, 0.16733066737651825]

start = robot_config.Tx("EE", feedback["q"], x= np.array(base))
result: start is:  [-0.33954067  1.79179413 -8.71078803]
´´´

studywolf, 2020-06-12:
Hello! Ah, this is because the `xyz_offset` parameter actually refers to an xyz offset _in task space_ (i.e. the hand is assumed to be at [0, 0, 0] in the last coordinate frame on the robot, specifying `xyz_offset` changes the hand location in the last coordinate frame), not in allocentric coordinates. 

So for your case instead of using that parameter you can just add the xyz position of the base of the arm to the output from `robot_config.Tx`.

```
start = robot_config.Tx("EE", feedback["q"]) + np.array(base)
```
should give you what you want, I believe.

quartzolit, 2020-06-12:
And in case for osc.generate scenario? What should I do to repair this problem of xyz_offset?

studywolf, 2020-06-12:
Hm, I'm sorry I need a bit more information. Do you mean the target coordinates need to have the rover position subtracted before being passed into `osc.generate`? My guess is that you don't require the `xyz_offset` parameter.

quartzolit, 2020-06-12:
As far as I know, I need to tell the coordinates of the base of the arm, because inside the generate function is calculated the Tx and jacobian of each joint and link. If i don't pass any value of xyz_offset, it will be generated a distance of the end-effector and the target. But the end-effector will be calculated based on his base at [0,0,0]. And if the distance is too high of the origin. The results of torque u goes insanely high.

```
base = interface.get_xyz("UR5_link0_visible")
result: Base is:  [-9.712021827697754, 1.791999101638794, 0.16733066737651825]

start = robot_config.Tx("EE", feedback["q"]) + np.array(base)
result: start is:  [-9.88423183  1.79179413  1.16856447] (this solved the first problem)

u = ctrlr.generate(q=feedback["q"], dq=feedback["dq"], target=target,)

result: u values are: [62.58568618 88.65189519 29.98363745  2.13804257 -7.78033134 -0.15512129]
```

studywolf, 2020-06-12:
I would guess that you also need to subtract the base location from the target position, because the target needs to be passed in in robot-centric coordinates not world coordinates. 
e.g.
```
u = ctrlr.generate(q=feedback['q'], dq=feedback['dq'], target=target=np.array(base))
```

quartzolit, 2020-06-12:
Oh...Got it now. Sorry by my misunderstanding. Now it makes perfect sense

studywolf, 2020-06-12:
No problem! Happy to help :)

closed 2020-06-12 by studywolf