abr/abr_control #81: Inverse kinematic for Mujoco

a-z-e-r-i-l-a, 2020-07-11:
Hi, I wanted to ask if your implementations also allow inverse kinematic for mujoco.

I am interested in getting joints' values (MjData.qpos) for a given the desired end-effector pose, and not necessarily wish to move the arm to that position (similar to PyBullet's inverse kinematic api if you're familiar with it).

studywolf, 2020-07-16:
Hello! 

One of the standard ways is to make a mocap object and weld it to the part you want to control, I've added some files that do this, for reference. 

[invkin_example.zip](https://github.com/abr/abr_control/files/4934128/invkin_example.zip)

It doesn't sound like this will address your use case, though. We do have an inverse kinematics planner, but it requires access to the transformation matrices of the robot, which is a feature not available currently in our Mujoco robot_config. I remember it not being straight-forward to find from the Mujoco API, but I'll look into it again as this would be a very nice feature to offer.

studywolf, 2020-07-17:
Hello! Had the chance to look at this today and was able to get it working. It should hopefully be merged into master soon, but if you check out the `m4` branch there is now a `docs/examples/Mujoco/path_planning_inverse_kinematics.py` file that you can run that I believe is an example of how to address your use case. If you don't need the whole path you can just access the last configuration in the path with `path_planner.position_path[-1]` once you've called the `generate` function to create the path.

studywolf, 2020-08-15:
Hopefully this addressed your issues, I'll close this for now and we can re-open if it's not working for you.

closed 2020-08-15 by studywolf