abr/abr_control #83: ask for  reference about the controllers

vanusy, 2020-08-07:
I have seen there are several different controllers in the "abr_control/abr_control/controllers/", such as OSC and sliding et al. Could you give me some references related to your controllers? Many thanks for your kindly help :-D

studywolf, 2020-08-11:
Hello! You're referring to the OSC, right? I've added one for that controller in our open PR, and expanded the reference for the sliding controller. If there are any other questions about the others I can try to track down some references for those too! The references for sliding and OSC are 

OSC: [Khatib, Oussama. "A unified approach for motion and force control of robot manipulators: The operational space formulation." IEEE Journal on Robotics and Automation 3.1 (1987): 43-53.]
Sliding: [Slotine, Jean-Jacques E., and Weiping Li. "On the adaptive control of robot manipulators." The international journal of robotics research 6.3 (1987): 49-59.]

closed 2020-08-11 by studywolf

vanusy, 2020-08-12:
Yes, I'm referring to the OSC. Thanks a lot for your help. :-)

studywolf, 2020-08-12:
You're welcome!