abr/abr_control #97: MujocoConfig Error

jeanbuntu, 2022-05-03:
AttributeError: type object 'MujocoConfig' has no attribute 'Config'

```
from abr_control.arms.mujoco_config import MujocoConfig as arm

# 'ur5', 'onelink', 'twolink', and 'threelink' also available as arm_model's
arm_model = 'jaco2'
robot_config = arm.Config(arm_model)
```

p3jawors, 2022-05-03:
Please take a look at the Mujoco examples in the `docs/examples/Mujoco` folder. You'll have several examples on how to use the arm configs in Mujoco with various path planners and controllers. For your above issue, you would instantiate the `robot_config` as...
```
from abr_control.arms.mujoco_config import MujocoConfig as arm

# 'ur5', 'onelink', 'twolink', and 'threelink' also available as arm_model's
arm_model = 'jaco2'
robot_config = arm(arm_model)
```
Cheers

studywolf, 2022-05-03:
It seems that we have the wrong sample code in the README, thanks for bringing this to our attention @jeanbuntu ! We'll correct that, and in the mean time yes, as @p3jawors mentions, please refer to the example code in the `docs/examples/Mujoco` folder. We'll also make sure to reference that in the README.

jeanbuntu, 2022-05-03:
Understood, will do. Appreciate the time and response.