abr/abr_control #108: Non-linear trajectory issue

zxwang96, 2023-04-26:
Hi,

I'm using the attached mjcf to play with the `examples/Mujoco/force_osc_xyz_linear_path_gaussian_velocity.py `. But I found the trajectory of the end-effector is not linear as the figure shows. My questions are:

1. Is there anyway to fix this issue?
2. If it's impossible, I was wondering if you know any robotiq85 gripper working fine with the built-in ur5 of abr_control?
3. May I know the metrics used in this project?

eef traj:
[image]

mjcf xml:
[ur5e_gripper.zip](https://github.com/abr/abr_control/files/11337055/ur5e_gripper.zip)

Thank you!

p3jawors, 2023-04-27:
Hi,

When I run the sim the reach looks quite straight in its trajectory. It looks like it may be a plotting artefact. If you set your axes to be equal it should look better, if you look at your one axis the range is much smaller than the others.

If you add this:
`ax3.set_aspect('equal')`
before the `plt.show()` call you should get the following figure:
[image]

Let me know if this doesn't resolve the issue.

Cheers,

Pawel

closed 2023-04-27 by p3jawors

zxwang96, 2023-04-27:
Hi Pawel, thank you for the answer. I don't think it's an artifact issue. Instead, I found if I call `pp.next()` for each 2 iterations in the loop, the problem resolved. However, the trajectory(blue) is a little bit off from the planned path(yellow) as the image shows. Could you provide any insight? And I'd like to double check if you are using the mjcf file I attached in last initial post? Thank you!
[image]

quangminh2702, 2023-05-09:
hi @zxwang96. Could you please send me the config.py file for the UR5e robot? Thank you!

quangminh2702, 2023-05-09:
Hi @zxwang96. I am also trying to control the **ur5e** robot, but I do not have the `config.py` for that. So it would be great if you could send me the config file for the ur5e robot. Thank you!

zxwang96, 2023-05-09:
which config.py? I just use the mjcf file downloaded from mujoco_menagerie and modify the `examples/Mujoco/force_osc_xyz_linear_path_gaussian_velocity.py` file accordingly