abr/abr_control #110: Control of two robots

quangminh2702, 2023-06-14:
Thank you for your useful repo. We are trying to perform the assembly experiments. Is it possible to control two robots at the same time?

p3jawors, 2024-09-25:
Hello @quangminh2702,

At the moment we do not support connecting multiple robots with either CoppeliaSim or Mujoco. From a cursory search it looks like [this could be possible in Mujoco](https://www.roboti.us/forum/index.php?threads/simulation-multiple-robot-in-one-scene.3743/), but we have no plans at the moment of updating the features. If you do add this feel free to make a PR and we could look into merging it.

Cheers

Tadinu, 2024-09-25:
https://github.com/ir-lab/irl_control